gawindx/WinNUT-Client #126: Windows 11

ChuckRider, 2022-02-16:
I am trying to get the client to work on Windows 11.  It runs fine on Windows 10.  When I launch the app I get this:

WinNUT Bug Report
Os Version : 6.2.9200.0
WinNUT Version : 2.1.7740.35837

WinNUT Parameters : 
{
  "Delay": 5000,
  "AutoReconnect": "True",
  "MinInputVoltage": 110,
  "MaxInputVoltage": 130,
  "FrequencySupply": 1,
  "MinInputFrequency": 55,
  "MaxInputFrequency": 65,
  "MinOutputVoltage": 110,
  "MaxOutputVoltage": 130,
  "MinUPSLoad": 0,
  "MaxUPSLoad": 100,
  "MinBattVoltage": 6,
  "MaxBattVoltage": 18,
  "MinimizeToTray": "False",
  "MinimizeOnStart": "False",
  "CloseToTray": "False",
  "StartWithWindows": "True",
  "UseLogFile": "True",
  "Log Level": 3,
  "ShutdownLimitBatteryCharge": 30,
  "ShutdownLimitUPSRemainTime": 120,
  "ImmediateStopAction": "False",
  "Follow_FSD": "False",
  "TypeOfStop": 0,
  "DelayToShutdown": 15,
  "AllowExtendedShutdownDelay": "False",
  "ExtendedShutdownDelay": 15,
  "VerifyUpdate": "True",
  "VerifyUpdateAtStart": "False",
  "DelayBetweenEachVerification": 2,
  "StableOrDevBranch": 1,
  "LastDateVerification": "2/15/2022 8:40:32 PM"
}

Error Message : 
Exception type: System.NullReferenceException
Exception message: Object reference not set to an instance of an object.
Exception stack trace: 
   at WinNUT_Client_Common.Nut_Socket.get_IsKnownUPS(String Test_UPSname)
   at WinNUT_Client_Common.UPS_Device.Connect_UPS()
   at WinNUT_Client_Common.UPS_Device..ctor(Nut_Parameter Nut_Config, Logger& LogFile)
   at WinNUT_Client.WinNUT.WinNUT_Load(Object sender, EventArgs e)
   at System.EventHandler.Invoke(Object sender, EventArgs e)
   at System.Windows.Forms.Form.OnLoad(EventArgs e)
   at System.Windows.Forms.Form.OnCreateControl()
   at System.Windows.Forms.Control.CreateControl(Boolean fIgnoreVisible)
   at System.Windows.Forms.Control.CreateControl()
   at System.Windows.Forms.Control.WmShowWindow(Message& m)
   at System.Windows.Forms.Control.WndProc(Message& m)
   at System.Windows.Forms.ScrollableControl.WndProc(Message& m)
   at System.Windows.Forms.Form.WmShowWindow(Message& m)
   at System.Windows.Forms.Form.WndProc(Message& m)
   at System.Windows.Forms.Control.ControlNativeWindow.OnMessage(Message& m)
   at System.Windows.Forms.Control.ControlNativeWindow.WndProc(Message& m)
   at System.Windows.Forms.NativeWindow.Callback(IntPtr hWnd, Int32 msg, IntPtr wparam, IntPtr lparam)

Last Events :
2/16/2022 10:36:23 AM Pid: 10192 WinNUT : Initialisation Globals Variables Complete
2/16/2022 10:36:23 AM Pid: 10192 WinNUT : Initialisation Params Complete
2/16/2022 10:36:23 AM Pid: 10192 WinNUT : Loaded Params Complete
2/16/2022 10:36:23 AM Pid: 10192 WinNUT : NotifyIcons Initialised
2/16/2022 10:36:23 AM Pid: 10192 WinNUT : Windows 10 Toast Notification Not Available. Too Old Windows Version
2/16/2022 10:36:23 AM Pid: 10192 UPS_Device : TCP Socket Created
2/16/2022 10:36:23 AM Pid: 10192 UPS_Device : TCP Socket Deconnected
2/16/2022 10:36:23 AM Pid: 10192 UPS_Device : Reconnection Process Started

ianhyzy, 2022-10-07:
I'm also not able to open the app on Windows 11:

WinNUT Bug Report
Os Version : 6.2.9200.0
WinNUT Version : 2.1.7740.35837

WinNUT Parameters : 
{
  "Delay": 0,
  "AutoReconnect": "True",
  "MinInputVoltage": 210,
  "MaxInputVoltage": 270,
  "FrequencySupply": 0,
  "MinInputFrequency": 40,
  "MaxInputFrequency": 60,
  "MinOutputVoltage": 210,
  "MaxOutputVoltage": 250,
  "MinUPSLoad": 0,
  "MaxUPSLoad": 100,
  "MinBattVoltage": 6,
  "MaxBattVoltage": 18,
  "MinimizeToTray": "False",
  "MinimizeOnStart": "False",
  "CloseToTray": "False",
  "StartWithWindows": "True",
  "UseLogFile": "False",
  "Log Level": 0,
  "ShutdownLimitBatteryCharge": 30,
  "ShutdownLimitUPSRemainTime": 120,
  "ImmediateStopAction": "False",
  "Follow_FSD": "False",
  "TypeOfStop": 0,
  "DelayToShutdown": 15,
  "AllowExtendedShutdownDelay": "False",
  "ExtendedShutdownDelay": 15,
  "VerifyUpdate": "False",
  "VerifyUpdateAtStart": "False",
  "DelayBetweenEachVerification": 2,
  "StableOrDevBranch": 0,
  "LastDateVerification": ""
}

Error Message : 
Exception type: System.ArgumentOutOfRangeException
Exception message: Value '0' is not a valid value for Interval. Interval must be greater than 0.
Parameter name: Interval
Exception stack trace: 
   at System.Windows.Forms.Timer.set_Interval(Int32 value)
   at WinNUT_Client.WinNUT.WinNUT_Load(Object sender, EventArgs e)
   at System.EventHandler.Invoke(Object sender, EventArgs e)
   at System.Windows.Forms.Form.OnLoad(EventArgs e)
   at System.Windows.Forms.Form.OnCreateControl()
   at System.Windows.Forms.Control.CreateControl(Boolean fIgnoreVisible)
   at System.Windows.Forms.Control.CreateControl()
   at System.Windows.Forms.Control.WmShowWindow(Message& m)
   at System.Windows.Forms.Control.WndProc(Message& m)
   at System.Windows.Forms.ScrollableControl.WndProc(Message& m)
   at System.Windows.Forms.Form.WmShowWindow(Message& m)
   at System.Windows.Forms.Form.WndProc(Message& m)
   at System.Windows.Forms.Control.ControlNativeWindow.OnMessage(Message& m)
   at System.Windows.Forms.Control.ControlNativeWindow.WndProc(Message& m)
   at System.Windows.Forms.NativeWindow.Callback(IntPtr hWnd, Int32 msg, IntPtr wparam, IntPtr lparam)

Last Events :
10/7/2022 3:21:00 PM Pid: 24268 WinNUT : Initialisation Globals Variables Complete
10/7/2022 3:21:00 PM Pid: 24268 WinNUT : Initialisation Params Complete
10/7/2022 3:21:00 PM Pid: 24268 WinNUT : Loaded Params Complete
10/7/2022 3:21:00 PM Pid: 24268 WinNUT : NotifyIcons Initialised
10/7/2022 3:21:00 PM Pid: 24268 WinNUT : Windows 10 Toast Notification Not Available. Too Old Windows Version
10/7/2022 3:21:00 PM Pid: 24268 UPS_Device : TCP Socket Created
10/7/2022 3:21:00 PM Pid: 24268 UPS_Device : Enter GetUPSVar
10/7/2022 3:21:00 PM Pid: 24268 UPS_Device : Process Result With ups.mfr : VAR ups ups.mfr "CPS"
10/7/2022 3:21:00 PM Pid: 24268 UPS_Device : Enter GetUPSVar
10/7/2022 3:21:00 PM Pid: 24268 UPS_Device : Process Result With ups.model : VAR ups ups.model "OR1500PFCRT2U"
10/7/2022 3:21:00 PM Pid: 24268 UPS_Device : Enter GetUPSVar
10/7/2022 3:21:00 PM Pid: 24268 UPS_Device : Apply Fallback Value when retrieving ups.serial
10/7/2022 3:21:00 PM Pid: 24268 UPS_Device : Enter GetUPSVar
10/7/2022 3:21:00 PM Pid: 24268 UPS_Device : Apply Fallback Value when retrieving ups.firmware
10/7/2022 3:21:00 PM Pid: 24268 UPS_Device : Enter GetUPSVar
10/7/2022 3:21:00 PM Pid: 24268 UPS_Device : Apply Fallback Value when retrieving battery.capacity
10/7/2022 3:21:00 PM Pid: 24268 UPS_Device : Enter GetUPSVar
10/7/2022 3:21:00 PM Pid: 24268 UPS_Device : Apply Fallback Value when retrieving output.frequency.nominal

gbakeman, 2022-10-14:
@ianhyzy It looks like your error is a known problem and not actually related to Windows 11. You can find a debug build released at the new repository here: nutdotnet/WinNUT-Client/issues/35. @ChuckRider please also check out the new repository and see if your issue isn't fixed. Feel free to close this issue in the meantime.