gawindx/WinNUT-Client #132: Will not re-establish connection after sleep mode Windows 11

conman33158, 2022-04-08:
Hi all, I’m getting the following on Windows 11, the WinNUT program will not reconnect after switching on after sleep mode even though the “re-establish connection” box in ticked in the connection settings.

[Id 2: 08/04/2022 11:25:56] Windows standby. WinNUT will disconnect.
[Id 3: 08/04/2022 11:26:16] Restarting WinNUT after waking up from Windows.
[Id 4: 08/04/2022 11:26:37] Connection to Nut Host 192.168.0.55:3493 Failed: Connection Error

I manually have to go to connection and hit “reconnect”

Any fix for this or is there something I'm missing? By the way, love the program 👍

gbakeman, 2022-04-11:
I was going to say that this is a duplicate of #88 , but are you saying that bug was introduced between the stable and pre-release versions?

conman33158, 2022-04-11:
Yes my friend, exactly as above. I totally un-installed the pre-release version and removed from windows registry etc and then reboot the PC, after the install of "latest version" v2.0.7722.30975 it worked no problem. When windows awoke from sleep the connection was re-established between 15-30 seconds and has been flawless 3 days so far 👍