gawindx/WinNUT-Client #134: Bad German translation

ChrisBCH, 2022-04-19:
The German translation is really not good (to say it nicely) an in some cases misleading. My questions:
1) Can I switch to English UI (my Windows 11 is German)?
2) I can provide better translations, but I need to know the English original first in order to make the translation better. So basically same question: How can I get the English original?
3) Wouldn't it be better to let the user decide about the language? I would prefer English UI.

gbakeman, 2022-04-19:
Hello Chris,

I thought it was possible to change the language in app, but it appears not. I'd suggest changing the name of this issue to something like "Manual language selection" since that would be a good feature to have.

Please see PRs #108 and #131 as these are tracking German translation updates. We'd appreciate having your feedback on those.

In the meantime, this repository is currently in stasis and awaiting either @gawindx to return or a migration to a GitHub organization which I'm slowly working on at the moment.

TigraTT, 2022-08-08:
i also found some wrong translations and tried to change them, but i coulnd get it work.

clang88, 2023-05-12:
I was also quite shocked by the poor german translations in this app and the lack of options to change it. Of course, you could change the translations and build from source, but it is not ideal.

A workaround that somewhat works is just simply deleting the localization folders... This works fine if you run the .exe directly, however, if you use the shortcut created by the installer it tries to install something. A manual language selection would definitely be much appreciated.

gbakeman, 2023-05-12:
Everyone: this repository is no longer being maintained. Improved translations and more fixes can be found at https://github.com/nutdotnet/WinNUT-Client. If you see some translations that need improving, please consider [contributing to the project](https://github.com/nutdotnet/WinNUT-Client/wiki/Translations).