gawindx/WinNUT-Client #135: Bug Report: Beta v2.1.7740.35837 crashes

ChrisBCH, 2022-04-23:
Since I have disconnet issues with the latest stable release when going 'on battery', I tried the latest beta [v2.1.7740.35837].
This version crahes immediately when starting WinNut (on my German Windows 11).
Here is the crash report (full report attached):

Error Message : 
Exception type: System.ArgumentOutOfRangeException
Exception message: Der Text muss kürzer als 64 Zeichen sein.
Parametername: Text
Der tatsächliche Wert war WinNUT-Client - 2.1
In Verbindung gebracht
Online
Batterie OK.
Exception stack trace: 
   bei System.Windows.Forms.NotifyIcon.set_Text(String value)
   bei WinNUT_Client.WinNUT.Event_UpdateNotifyIconStr(String Reason, String Message)
   bei WinNUT_Client.WinNUT.Update_UPS_Data()
   bei WinNUT_Client.WinNUT.Retrieve_UPS_Datas(Object sender, EventArgs e)
   bei System.Windows.Forms.Timer.OnTick(EventArgs e)
   bei System.Windows.Forms.Timer.TimerNativeWindow.WndProc(Message& m)
   bei System.Windows.Forms.NativeWindow.Callback(IntPtr hWnd, Int32 msg, IntPtr wparam, IntPtr lparam
[WinNUTCrashReport.txt](https://github.com/gawindx/WinNUT-Client/files/8546652/WinNUTCrashReport.txt)
)

gbakeman, 2022-04-24:
I believe this is related to #114, #81, and #74.

nibbsification, 2022-06-23:
I'm having the same problem. Tried latest beta, uninstallung and rebooting before installing the beta also didn't help. 
Also renaming or deleting the German .dll did not help, as it's recreated at startup.

gbakeman, 2022-08-07:
@nibbsification please follow our latest issue at https://github.com/nutdotnet/WinNUT-Client/issues/5, there's a new pre-release that should fix this issue. Please provide your feedback there.