gawindx/WinNUT-Client #136: NEW Repository Location

gbakeman, 2022-05-05:
I'm not sure what happened to our repository maintainer @gawindx, but I know there are some important bugs and other enhancements that people have been patiently waiting for so we need to start getting builds out.

I've created a new [Github Organization](https://github.com/nutdotnet), and I've imported the WinNUT-Client project [here](https://github.com/nutdotnet/WinNUT-Client). **Please direct all new issues and pull requests to that project.** I plan on being semi-active and approving PRs and hopefully pushing out a build or two in the near future.

deluxghost, 2022-05-16:
you can also import this issue: #88

FileCity, 2022-07-26:
As a personal perception, is there a plan to translate this project to C# for better maintainability ?
This can be done with automated tools as VB.NET and C# use the same common language runtime.
I don't have anything against VB.NET but this language, according to me, is less future proof.

gbakeman, 2022-07-27:
@FileCity That's a great question, and I think it's important we discuss the future of WinNUT. I personally prefer C# and I've been working on a library for WinNUT writte in C#, so program logic can be extracted as much as possible. I would like to see WinNUT become a multi-platform program (maybe think of a different name), as well as have a modern interface (.Net MAUI was released recently, worth investigating). I think it would be a good idea to move this to a discussion on the new repository.

gbakeman, 2022-07-27:
@deluxghost Also want to mention that issue is now live here: https://github.com/nutdotnet/WinNUT-Client/issues/4