gawindx/WinNUT-Client #138: WinNUT-Client won't even start due to some sort of settings issue

26tajeen, 2022-06-15:
`Exception type: System.ArgumentOutOfRangeException
Exception message: Value '0' is not a valid value for Interval. Interval must be greater than 0.
Parameter name: Interval
Exception stack trace: 
   at System.Windows.Forms.Timer.set_Interval(Int32 value)
   at WinNUT_Client.WinNUT.WinNUT_Load(Object sender, EventArgs e)
   at System.EventHandler.Invoke(Object sender, EventArgs e)
   at System.Windows.Forms.Form.OnLoad(EventArgs e)
   at System.Windows.Forms.Form.OnCreateControl()
   at System.Windows.Forms.Control.CreateControl(Boolean fIgnoreVisible)
   at System.Windows.Forms.Control.CreateControl()
   at System.Windows.Forms.Control.WmShowWindow(Message& m)
   at System.Windows.Forms.Control.WndProc(Message& m)
   at System.Windows.Forms.ScrollableControl.WndProc(Message& m)
   at System.Windows.Forms.Form.WmShowWindow(Message& m)
   at System.Windows.Forms.Form.WndProc(Message& m)
   at System.Windows.Forms.Control.ControlNativeWindow.OnMessage(Message& m)
   at System.Windows.Forms.Control.ControlNativeWindow.WndProc(Message& m)
   at System.Windows.Forms.NativeWindow.Callback(IntPtr hWnd, Int32 msg, IntPtr wparam, IntPtr lparam)`

There's no way to clear settings and restart.

traceon, 2022-07-03:
Same for me.

traceon, 2022-07-03:
Changing `Computer\HKEY_CURRENT_USER\Software\WinNUT\Connexion\Delay` registry `DWORD` value to `1`, and making sure `Computer\HKEY_CURRENT_USER\Software\WinNUT\Connexion\ServerAddress` value is an IP address and not a host name, fixed the crash for me.