gawindx/WinNUT-Client #140: WinNut Immediately disconnects from UPS-Server upon the UPS going on battery

MartinKurtz, 2022-06-24:
Basically as soon as the UPS goes on battery power, the WinNUT client loses connection, log file is attached. I suspect this is related to issue nr 114 and some others.
[WinNUT-client-2022-06-24 - Kopie.txt](https://github.com/gawindx/WinNUT-Client/files/8977063/WinNUT-client-2022-06-24.-.Kopie.txt)
.

ChrisBCH, 2022-08-04:
I am having the same issue.

MartinKurtz, 2022-08-04:
Yes:

I have not tested this, but you can possibly skip 1 to 4 and use the "latest" version and do the same thing

1. Uninstall WinNUT
2. Resart PC
3. Install the newest beta release
4. It should now crash when starting it up saying something about string length
5. Close/Shutdown WinNUT
6. Go to whereever you installed WinNUT
7. Enter a folder named de-DE for german,  if you have another language, pick that folder
8. Delete the .dll file in there and start WinNUT
9. WinNUT should now fall back to the default language, which is english, but if you can read this or are computerlingual in any way this shouldnt be a problem, and in that language the string that is apparently the issue, is shorter and does apparently not cause this issue usually
10. Magic

After following these steps it worked fine for me, however just installing the newest pre release version (https://github.com/gawindx/WinNUT-Client/releases/tag/2.1.7740.35837) at the time of writing did not fix the issue even though it said so in the changelog

MartinKurtz, 2022-08-04:
I also deliberately did not close the issue yet, as the solution was a workaround

ChrisBCH, 2022-08-06:
Thanks a lot for your explanation!
Installation according to your steps described above worked wonderfully, and using English as UI is fine for me.
WinNut also detected when the UPS went on battery supply.
But then a strange thing happened: The UPS completely switched off. This never happened to me before and I have no idea what caused this.
So in summary I am a bit confused now. Was this coincidence or is it related to the Beta WinNut?

gbakeman, 2022-08-06:
@ChrisBCH I'm not quite sure why your UPS switched off, I don't think WinNUT is supposed to control it in that way. I've released a pre-release [here](https://github.com/nutdotnet/WinNUT-Client/releases/tag/v2.2.8253), please give it a try and let me know if you still have the same problem as above. If so, then please make sure you upload the full debug log to the **new** issue [here](https://github.com/nutdotnet/WinNUT-Client/issues/5).

efebert, 2024-06-22:
I'm new in github as I have need for this project.
My installation is:
- APC Back UPS XS 700U
- Synology DS418play, DSM 7.1.1-42962 Update 6 (actual)
- Win 11 Home, language german (actual)
- WinNUT client 2.0.7722.30975 (stable, not BETA)
- 
I also discovered this problem and applied the workaround from MartinKurz above.
It works for me fine; shutdown is performed as set up

gbakeman, 2024-06-22:
Hi @efebert ,

I'm glad you found a solution that helped here. I want to mention that this specific project on GitHub is no longer maintained, and I've moved it over to [here](https://github.com/nutdotnet/WinNUT-Client). There have been several updates since the latest release here which I think you may be interested in. In the future, feel free to create a discussion there and I'll be happy to help.

efebert, 2024-06-22:
Hi @gbakeman,

thank you for this notification.
I switched to the project pointed to by you and installed it.
Everythging works fine, in german language, too.
In future, I'll follow the new project.