gawindx/WinNUT-Client #141: Will not shutdown no matter what I try

dmudgeau, 2022-06-29:
hi 'm new here and got Win Nut working on Windows 10 talking to my CyberPower VP1600ELCD running on latest Nut server on my pi.. all good

Problem ism, no matter what settings I use I am not getting Windows to shutdown. I have tried shutdown and hibernate. I have the following settings:
Shutdown if battery lower than 98
runtime lower than 3600

I have tried both immediate stop action and delayed. I have shutdown on FSD enabled

Nothing every happens! Win Nut correctly changes to on battery, I look at the UPS variables and remaining time is there. I note that remaining time is NULL on the UI but thats seems to be a feature

So not sure what I am doing wrong! Nothing happens on battery it was going to like 70% and still nothing. Do I have to run it in admin mode? Logs dont tell me anything except UPS events - on battery, on AC etc

Thanks

metheofanis, 2023-06-22:
Same problem here. Any solution yet?

gbakeman, 2023-06-22:
Everyone,

This repository is no longer maintained. Development has moved to https://github.com/nutdotnet/WinNUT-Client. Please download the latest version from there and open a bug report if you're still experiencing the issue.

@dmudgeau If you could please close this ticket, hopefully people will be more likely to see that this repository is inactive.