gawindx/WinNUT-Client #142: Store configuration for "all users"

fatbasstard, 2022-07-05:
At the moment the settings are stored in the `HKEY_CURRENT_USER`, this is annoying when using automation to install and configure WinNUT.

We use Chef to install and configure WinNUT which works great if I manually kick off a Chef run, but when it runs in the automated scheduled way it's not useable: The schuled task for Chef runs as `SYSTEM` so it's no so trivial to set the correct settings.

Is it an option to move the configuration to `HKEY_LOCAL_MACHINE` or maybe even better support both (let CURRENT_USER overwrite LOCAL_MACHINE)?

gbakeman, 2022-07-05:
I opened issue #105 to debate a future system for saving configuration & settings, please take a look if you have any ideas. I personally don't believe that the Registry is a feasible location for data storage for the long-term future of this project. My suggestion is that we investigate various formats of file-based storage, but my instinct would be to make this specific to the user. It sounds like you want a global (system-wide) settings system, could you explain your case a little more? I'm not familial with Chef.

fatbasstard, 2022-07-08:
Chef is a configuration management tool. In short: Automated desired state maintenance. Other tools usually used are Ansible or Puppet.

In short: I've got a bunch of Windows machines that are maintained through code and automation. Chef runs under the `SYSTEM` user and (in short) installs and configured the machines and software.

gbakeman, 2022-07-08:
If chef allows for deploying application state configurations to the user (directory), then I think this may be the best solution going forward. Again I'm not familiar with Chef or conf. management in general so I'll need your expertise on that matter.