gawindx/WinNUT-Client #148: WinNUT doesn't reconnect to UPS on resume from sleep

cinergi2, 2022-07-31:
On Windows 11 (possibly also Windows 10, haven't tried it), when my PC resumes from sleep, WinNUT Client sometimes doesn't reconnect to the UPS server.  I have to manually select Connection > Reconnect, then it's fine.

gbakeman, 2022-08-01:
It looks like this is a duplicate of #132 and #88 . Can you please verify what version of WinNUT you're running? I'm tracking this over at the new repository, https://github.com/nutdotnet/WinNUT-Client/issues/4. Please watch that issue for updates.

cinergi2, 2022-08-01:
Thanks.  It does indeed seem to be a duplicate of the referenced issues; sorry, I was unaware of them.  I'm running version v2.1.7740.35837.