gawindx/WinNUT-Client #150: Increasing battery max in Calibration extends the gauge instead of adjusting scale.

hugalafutro, 2022-08-07:
I hope the title makes sense. I am using version v2.1.7740.35837 on Windows 11 x64 as netclient for nut server running on pi4.

I increased the max to 28 as my charged battery sits at around 27.1 and expected the gauge to remain same size and the data on it to "shrink", instead the gauge grew a new notch triggering my imagined ocd how it's all uneven now.

I don't think the needle points at correct place either.

[image]

Otherwise everything seems to work ok, thanks for your work on this tool!

gbakeman, 2022-08-07:
@hugalafutro We've moved the repository to a new location, would you mind duplicating this over at [nutdotnet/WinNUT-Client](https://github.com/nutdotnet/WinNUT-Client/issues/new/choose) and I'll take a look?

hugalafutro, 2022-08-08:
done @ https://github.com/nutdotnet/WinNUT-Client/issues/18

closed 2022-08-08 by hugalafutro