gawindx/WinNUT-Client #157: String was not recognized as a valid DateTime

quantass, 2022-11-06:
OS: Windows 11
Locale: Canada

Exception type: System.FormatException
Exception message: String was not recognized as a valid DateTime.
Exception stack trace: 
   at System.DateTimeParse.Parse(String s, DateTimeFormatInfo dtfi, DateTimeStyles styles)
   at System.Convert.ToDateTime(String value, IFormatProvider provider)
   at System.String.System.IConvertible.ToDateTime(IFormatProvider provider)
   at System.Convert.ToDateTime(Object value)
   at WinNUT_client.Update_Gui.VerifyUpdate()
   at WinNUT_client.Update_Gui.SetVisibleCore(Boolean value)
   at System.Windows.Forms.Control.set_Visible(Boolean value)
   at WinNUT_client.WinNUT.WinNUT_Load(Object sender, EventArgs e)
   at System.EventHandler.Invoke(Object sender, EventArgs e)
   at System.Windows.Forms.Form.OnLoad(EventArgs e)
   at System.Windows.Forms.Form.OnCreateControl()
   at System.Windows.Forms.Control.CreateControl(Boolean fIgnoreVisible)
   at System.Windows.Forms.Control.CreateControl()
   at System.Windows.Forms.Control.WmShowWindow(Message& m)
   at System.Windows.Forms.Control.WndProc(Message& m)
   at System.Windows.Forms.ScrollableControl.WndProc(Message& m)
   at System.Windows.Forms.ContainerControl.WndProc(Message& m)
   at System.Windows.Forms.Form.WmShowWindow(Message& m)
   at System.Windows.Forms.Form.WndProc(Message& m)
   at System.Windows.Forms.Control.ControlNativeWindow.OnMessage(Message& m)
   at System.Windows.Forms.Control.ControlNativeWindow.WndProc(Message& m)
   at System.Windows.Forms.NativeWindow.Callback(IntPtr hWnd, Int32 msg, IntPtr wparam, IntPtr lparam)

gbakeman, 2022-11-06:
Hi,

This repository is no longer being maintained, and development has moved to https://github.com/nutdotnet/WinNUT-Client. Please try the latest pre-release version and see if that is any help. If not, then please open a new issue with as many details, including logs and things you've tried and I'll see if I can help.

HotelAlpha4, 2023-02-13:
If it helps anyone else, I got this issue because I had set the short date format in Windows to one of two short date formats that make sense: yyyy-MM-dd or yyyy/mm/dd. yy/MM/dd also fails, but no one would use that date format, surely! That makes even less sense than MM/dd/yy...

Set it to any other date format, and WinNut works fine.