gawindx/WinNUT-Client #162: No auto reconnect after connection failure

SSPW, 2022-12-08:
ver 2.0.7722.30975 on win10

Works fine, until server is lost, ie restart nut server or a temporary break in network between server and winnut etc

winnut does not appear to make any effort to reconnect automatically.

Manually using Connection -> Reconnect works.

I have monitored the conversation with wireshark. The winnut client seems to make no attempt to recconnect automatically after the connection is broken. Again manual using reconnect from menu works fine.

gbakeman, 2022-12-08:
Thank you for the detailed troubleshooting. This repository is no longer maintained, and development has moved to https://github.com/nutdotnet/WinNUT-Client. Please try the latest pre-release version and see if that is any help.

SSPW, 2022-12-08:
My bad! Ill try the new repo.