gawindx/WinNUT-Client #167: No longer starts on Win10, used to work, no changes

myself248, 2023-04-30:
WinNUT Bug Report
Os Version : 6.2.9200.0
WinNUT Version : 2.1.7740.35837

WinNUT Parameters : 
{
  "Delay": 5000,
  "AutoReconnect": "True",
  "MinInputVoltage": 90,
  "MaxInputVoltage": 140,
  "FrequencySupply": 1,
  "MinInputFrequency": 55,
  "MaxInputFrequency": 65,
  "MinOutputVoltage": 90,
  "MaxOutputVoltage": 140,
  "MinUPSLoad": 0,
  "MaxUPSLoad": 100,
  "MinBattVoltage": 20,
  "MaxBattVoltage": 30,
  "MinimizeToTray": "False",
  "MinimizeOnStart": "False",
  "CloseToTray": "False",
  "StartWithWindows": "True",
  "UseLogFile": "True",
  "Log Level": 0,
  "ShutdownLimitBatteryCharge": 50,
  "ShutdownLimitUPSRemainTime": 600,
  "ImmediateStopAction": "True",
  "Follow_FSD": "False",
  "TypeOfStop": 0,
  "DelayToShutdown": 15,
  "AllowExtendedShutdownDelay": "False",
  "ExtendedShutdownDelay": 15,
  "VerifyUpdate": "False",
  "VerifyUpdateAtStart": "False",
  "DelayBetweenEachVerification": 2,
  "StableOrDevBranch": 0,
  "LastDateVerification": ""
}

Error Message : 
Exception type: System.NullReferenceException
Exception message: Object reference not set to an instance of an object.
Exception stack trace: 
   at WinNUT_Client_Common.Nut_Socket.get_IsKnownUPS(String Test_UPSname)
   at WinNUT_Client_Common.UPS_Device.Connect_UPS()
   at WinNUT_Client_Common.UPS_Device..ctor(Nut_Parameter Nut_Config, Logger& LogFile)
   at WinNUT_Client.WinNUT.WinNUT_Load(Object sender, EventArgs e)
   at System.EventHandler.Invoke(Object sender, EventArgs e)
   at System.Windows.Forms.Form.OnLoad(EventArgs e)
   at System.Windows.Forms.Form.OnCreateControl()
   at System.Windows.Forms.Control.CreateControl(Boolean fIgnoreVisible)
   at System.Windows.Forms.Control.CreateControl()
   at System.Windows.Forms.Control.WmShowWindow(Message& m)
   at System.Windows.Forms.Control.WndProc(Message& m)
   at System.Windows.Forms.ScrollableControl.WndProc(Message& m)
   at System.Windows.Forms.Form.WmShowWindow(Message& m)
   at System.Windows.Forms.Form.WndProc(Message& m)
   at System.Windows.Forms.Control.ControlNativeWindow.OnMessage(Message& m)
   at System.Windows.Forms.Control.ControlNativeWindow.WndProc(Message& m)
   at System.Windows.Forms.NativeWindow.Callback(IntPtr hWnd, Int32 msg, IntPtr wparam, IntPtr lparam)

Last Events :
4/30/2023 2:52:41 PM Pid: 10456 WinNUT : Initialisation Globals Variables Complete
4/30/2023 2:52:41 PM Pid: 10456 WinNUT : Initialisation Params Complete
4/30/2023 2:52:41 PM Pid: 10456 WinNUT : Loaded Params Complete
4/30/2023 2:52:41 PM Pid: 10456 WinNUT : NotifyIcons Initialised
4/30/2023 2:52:41 PM Pid: 10456 WinNUT : Windows 10 Toast Notification Not Available. Too Old Windows Version
4/30/2023 2:52:41 PM Pid: 10456 UPS_Device : TCP Socket Created
4/30/2023 2:52:42 PM Pid: 10456 UPS_Device : TCP Socket Deconnected
4/30/2023 2:52:42 PM Pid: 10456 UPS_Device : Reconnection Process Started

myself248, 2023-04-30:
I saw some other notes in similar issues about changing IP addresses or wireless/wired connection types, but nothing like that has changed since my setup worked. I didn't use this PC for a few months, today I started it up and WinNUT crashed on launch. I tried reinstalling and no change.

gbakeman, 2023-04-30:
Hello,

Development has moved to [nutdotnet/WinNUT-Client](https://github.com/nutdotnet/WinNUT-Client). Please download the latest version there and let us know in the issues section if you're still experiencing the same problem. I'm working on several crash bugs right now, so your issue may be related if it wasn't fixed in a later release. Please also consider closing this issue so future users won't think this repository is still active.

myself248, 2023-04-30:
Closing accordingly. Thank you, will try that ASAP!

closed 2023-04-30 by myself248