gawindx/WinNUT-Client #169: Exception message: String was not recognized as a valid DateTime.

alebambi81, 2023-07-01:
Hello,
I've recently changed the regional format on my computer (from Italian to US) and WinNUT clways crashes at the startup with this error log:

Exception type: System.FormatException
Exception message: String was not recognized as a valid DateTime.
Exception stack trace: 
   at System.DateTimeParse.Parse(String s, DateTimeFormatInfo dtfi, DateTimeStyles styles)
   at System.Convert.ToDateTime(String value, IFormatProvider provider)
   at System.String.System.IConvertible.ToDateTime(IFormatProvider provider)
   at System.Convert.ToDateTime(Object value)
   at WinNUT_Client.Update_Gui.VerifyUpdate()
   at WinNUT_Client.Update_Gui.SetVisibleCore(Boolean value)
   at System.Windows.Forms.Control.set_Visible(Boolean value)
   at WinNUT_Client.WinNUT.WinNUT_Load(Object sender, EventArgs e)
   at System.EventHandler.Invoke(Object sender, EventArgs e)
   at System.Windows.Forms.Form.OnLoad(EventArgs e)
   at System.Windows.Forms.Form.OnCreateControl()
   at System.Windows.Forms.Control.CreateControl(Boolean fIgnoreVisible)
   at System.Windows.Forms.Control.CreateControl()
   at System.Windows.Forms.Control.WmShowWindow(Message& m)
   at System.Windows.Forms.Control.WndProc(Message& m)
   at System.Windows.Forms.ScrollableControl.WndProc(Message& m)
   at System.Windows.Forms.ContainerControl.WndProc(Message& m)
   at System.Windows.Forms.Form.WmShowWindow(Message& m)
   at System.Windows.Forms.Form.WndProc(Message& m)
   at System.Windows.Forms.Control.ControlNativeWindow.OnMessage(Message& m)
   at System.Windows.Forms.Control.ControlNativeWindow.WndProc(Message& m)
   at System.Windows.Forms.NativeWindow.Callback(IntPtr hWnd, Int32 msg, IntPtr wparam, IntPtr lparam)

The date has changed from dd/mm/yyy to mm/dd/yyy and the time from 24h to 12h.
I tried to uninstall and reinstall it but I always get the crash.
How can I reset or change the WinNUT's date and time format?

Thanks

Alessandro

gbakeman, 2023-07-01:
Hi Alessandro,

This repository is no longer maintained, and development has moved over to the new repository at https://github.com/nutdotnet/WinNUT-Client. Specifically, I believe this issue may be related to https://github.com/nutdotnet/WinNUT-Client/issues/78. Please add your feedback to that issue (this exception and any log you can provide), then close this issue so people know to go to the new repository.

Thank you!

alebambi81, 2023-07-01:
Thank you, I'll continue on the new repository!

closed 2023-07-01 by alebambi81