gawindx/WinNUT-Client #174: Monitor two (or more) UPS devices from WinNUT

rederikus, 2023-10-09:
I have two UPS devices.  One upstaris, one downstairs.  I use NUT to manage hem and WinNut to view them.  To do this I need to change WinNUT's Settings > Options Panel.

Is there any way to have either switchable UPS devices or a separate profile for each UPS?  Maybe some people have more that two UPSs.

Thanks

gbakeman, 2023-10-09:
Hi @rederikus ,

Development has moved to a new project, and specifically this issue: https://github.com/nutdotnet/WinNUT-Client/issues/90

I didn't think multi-UPS monitoring was going to happen for the current version of WinNUT, but I've been spending a lot of time on it so who knows. Subscribe to that issue/project for more info.

rederikus, 2023-10-10:
Thank you.  I am not tied to WinNut beyond that it was the only Windows app that I found to do the job.  I shall check out the other project immediately.

Thank you again.