gawindx/WinNUT-Client #175: Huw to switch to english UI on german windows?

stm64, 2023-11-05:
How can I switch to english UI? Readme says, that "WinNUT-Client V2 is natively multilingual, so it is no longer necessary to select your language from the software interface". In install directory there are resource DLL directories for de-DE, fr-FR etc. but no en-EN. How can I force WinNUT-Client to english UI?
Thanks, Stefan

gbakeman, 2023-11-05:
Hi, this project is no longer maintained and outdated. You can find the current project at https://github.com/nutdotnet/WinNUT-Client.

To answer your question, I think en-US is considered the native language in WinNUT. If you delete the language resource file WinNUT is using, then it should fallback to english. Having a language override may be a good idea - feel free to close this issue and open a new in the reposirory I linked, and I would be happy to look into it for you.

stm64, 2023-11-05:
Hi,
thanks for your fast feedback. I will switch to nutdotnet. Riding a dead horse makes no sense ;-)
But: I already tried to rename resource DLL or "de-DE" resource directory, but WinNUT-Client will reinstall this when starting. Also there is no en-US directory nor a (default/english) WinNUT-client.resources.dll in the install directory

gbakeman, 2023-11-05:
Just to be totally clear, WinNUT recreates the DLL when you start it? I had no idea, I would've thought it only does that when installing. The english language is probably embedded directly into WinNUT so I don't think it generates extra resource files. If deleting the extra language files doesn't help, this may require a feature to be implemented.

stm64, 2023-11-05:
Yes! I renamed "de-DE" to "de-DE.ignore". Now when I start WinNUT again it will not automatically reinstall "de-DE"!

In the meantime I switched to .NET-version. Behaviour is the same, it will re-install the missing "de-DE" directory.

BUT: if I "patch" WinNUT-Client.exe.manifest file and delete the corresponding dependency-entry

  <dependency>
    <dependentAssembly dependencyType="install" allowDelayedBinding="true" codebase="de-DE\WinNUT-Client.resources.dll" size="20992">
    </dependentAssembly>
  </dependency>

```
this re-install will not happen and WinNUT will "fall-back" to default language which is english!

BUT: I will stick to german, as with the .NET-version the translation to german is much better than in the old version. 

I like the idea of giving the user the option to select language when installing :

stm64, 2023-11-05:
Yes! I renamed "de-DE" to "de-DE.ignore". Now when I start WinNUT again it will not automatically reinstall "de-DE"!

In the meantime I switched to .NET-version. Behaviour is the same, it will re-install the missing "de-DE" directory on start-up.

BUT: if I "patch" WinNUT-Client.exe.manifest file and delete the corresponding dependency-entry for de-DE:

`  <dependency>
    <dependentAssembly dependencyType="install" allowDelayedBinding="true" codebase="de-DE\WinNUT-Client.resources.dll" size="20992">
      <assemblyIdentity name="WinNUT-Client.resources" version="2.2.8693.25940" language="de-DE" processorArchitecture="msil" />
      <hash>
        <dsig:Transforms>
          <dsig:Transform Algorithm="urn:schemas-microsoft-com:HashTransforms.Identity" />
        </dsig:Transforms>
        <dsig:DigestMethod Algorithm="http://www.w3.org/2000/09/xmldsig#sha256" />
        <dsig:DigestValue>BJ9ttXIMib7ECANPEuRFyefjze+yuvUJMPBOQ2ykqNo=</dsig:DigestValue>
      </hash>
    </dependentAssembly>
  </dependency>
`

this re-install will not happen and WinNUT will "fall-back" to default language which is english!

BUT: I will stick to german, as with the .NET-version the translation to german is much better than in the old version. 

I like the idea of giving the user the option to select language when installing.

gbakeman, 2023-11-05:
Ok, that's very interesting and I appreciate your work.

It definitely shouldn't be that difficult if you want to change the current language. I just opened a new issue that you can see above. Please subscribe and provide any feedback you might have. I decided to put it in the milestone tracking the next release _after_ the upcoming one (2.2.) WinNUT has been in a 2.2 pre-release state for what feels like forever, so I want to get bugs ironed out before making the official 2.2 release. But if you feel like this is an important feature that should be in 2.2, I'm open to reconsidering.

I'm also glad to hear that you find the newer German translations better. I know those were rough in the older versions of WinNUT, and we've had a lot of good work from several people. If you feel up to it, I'd love to have your time and edits if you see any area that needs it. Please take a look [here](https://github.com/nutdotnet/WinNUT-Client/wiki/Development) if that interests you.

Thanks again for your feedback. Feel free to close this issue and move conversation into the linked one above.