gawindx/WinNUT-Client #176: Export current configuration for import on another client?

stm64, 2023-11-05:
Is there a way to export working WinNUT-Client configuration on one PC for import on other PCs?

gbakeman, 2023-11-05:
Echoing my last comment about the repository being moved, but to answer this question: Currently, WinNUT uses the Windows Registry for storing settings. You can export your settings to a .reg file in two ways.

1. Open regedit.exe, and navigate to `HKEY_CURRENT_USER/SOFTWARE/WinNUT` and export that key, or
2. In a command window, run `reg export HKEY_CURRENT_USER/SOFTWARE/WinNUT .\export.reg` which will export to a file in the current directory.

stm64, 2023-11-05:
Frist I looked for a ups.ini on my system as this is the default file name when doing an import. But there is no such file on my PC. So I looked at registry and found HKCU/Software/WinNUT. But this cannot be used for direct import in another PC's registry as all nodes contain user's registry key like "S-1-5-21-xxxxxxx-xxxxxxxx-xxxxx-1000".

gbakeman, 2023-11-05:
Inside the WinNUT key should only be keys and values relating to WinNUT, is that not the case when you export the WinNUT key?

stm64, 2023-11-05:
No, unfortunatelly all keys contain this user specific key, even the WinNUT sub-keys like "Appearance", "Calibration" etc.

stm64, 2023-11-05:
BTW because I switched to .NET version I first uninstalled old WinNUT version. This uninstall did not remove the HKCU/Software/WinNUT branch in the registry!

As the .NET version uses the same names for registry this saved me from re-entering all the configuration settings. But normally I would expect uninstall to remove "everything".

gbakeman, 2023-11-05:
Good point about clearing out old data, that's not currently a feature in WinNUT and I'm honestly not sure if that will be a feature in the new Application Settings system or ClickOnce deployment that I want to use. I don't recall seeing it as an option in the current Setup project deployment system we use, so I don't think full data removal will be a feature at least in the near-future.

I am working on a migration to the standard .Net Framework Settings system in https://github.com/nutdotnet/WinNUT-Client/issues/100 if you want to follow along (would appreciate having some testers.) In the migration wizard, there will be an option to import settings from the Registry, export them to a .reg file on the filesystem, and also delete them from the Registry at the end. The user will also be able to ignore them/leave them alone, and start with default settings.

I apologize for my confusion as I've been in and out of the WinNUT registry tree quite a bit while working on this. I'm not sure I fully understand what you mean when you say there are nodes containing the user's registry key/GUID. I'm familiar with GUIDs like that and I know Windows uses them to identify user profiles/accounts, but I'm not sure why you're seeing it in WinNUT's key structure... that's a new one to me. Would you please post a screenshot of what you're seeing in the Windows Registry editor (partially censored if you wish?)

For reference, here's part of the key structure looks like for me:

[image]

```
Windows Registry Editor Version 5.00

[HKEY_CURRENT_USER\Software\WinNUT]

[HKEY_CURRENT_USER\Software\WinNUT\Appareance]
"MinimizeToTray"="True"
"MinimizeOnStart"="True"
"CloseToTray"="False"
"StartWithWindows"="True"

[HKEY_CURRENT_USER\Software\WinNUT\Calibration]
"MinInputVoltage"=dword:000000d2
"MaxInputVoltage"=dword:0000010e
"FrequencySupply"=dword:00000000
"MinInputFrequency"=dword:00000028
"MaxInputFrequency"=dword:0000003c
"MinOutputVoltage"=dword:000000d2
"MaxOutputVoltage"=dword:000000fa
"MinUPSLoad"=dword:00000000
"MaxUPSLoad"=dword:00000064
"MinBattVoltage"=dword:00000006
"MaxBattVoltage"=dword:00000012
```

stm64, 2023-11-06:
Shame on me, I looked in and exported WinNUT-branch from HKEY_USERS not HKEY_**CURRENT**_USER.
Apologies...

gbakeman, 2023-11-06:
No worries at all, sounds like you were seeing a copy of the current user's registry hive? Thank you again for the suggestions and feedback.