gawindx/WinNUT-Client #179: Minimize the Windows NUT Client by clicking the 'X' button instead of closing it entirely.

Bavarianspirit, 2023-12-04:
I believe it would be beneficial to have the option to minimize the Windows NUT Client when clicking the 'X' button, as opposed to closing it completely. This is particularly important because closing the software does not result in the shutdown of the Windows Client, as there is no background service associated with it.

gbakeman, 2023-12-04:
This repository is no longer maintained, and development has moved here: https://github.com/nutdotnet/WinNUT-Client

To answer your question, I'm pretty sure we have an option to make WinNUT minimize with the 'X' button, at least in the latest version. Please give it a try and open a new issue there if that's not the case.

Bavarianspirit, 2023-12-05:
Okay, thanks. Yes, that's what I meant.