gawindx/WinNUT-Client #181: Emulate upsmon slave behaviour

whc2001, 2024-01-20:
I have two computers attached to the same UPS. One of which running Linux acting as UPS server, another is a Windows desktop running WinNUT as UPS client. I configured the `upsd.users` as such:

```
[server]
    password = 123
    upsmon primary

[pc]
    password = 321
    upsmon secondary
```

and in `upsmon.conf` on the Linux machine: `MONITOR ups@localhost 1 server 123 primary`. After this, I can see in the `upsd` logs: `User server@127.0.0.1 logged into UPS [ups]`.

However, I can't see this line in the log when I configured WinNUT to the `pc` user. The connection is made and the monitoring values are displaying correctly, just that the remote `upsd` never knows there is another client relies on the same UPS. AFAIK this might cause some problems during the FSD sequence.

Reading the [documentation](https://networkupstools.org/docs/developer-guide.chunked/net-protocol.html#_login) seems like the client need to invoke `LOGIN <upsname>` after setting the username and password to let the `upsd` know. The user must have `upsmon` privilege configured in `upsd.users` or access will be denied. Before shutting down, invoke `LOGOUT` so the `upsd` knows that client has been turning off gracefully.

I think this should be nice to have, but for now I can't think of a proper way to add this functionality to WinNUT, like should we add a checkbox to enable this and show error if the account is denied due to no `upsmon` privilege configured, or we should try to login anyway and just do nothing if it's denied?

gbakeman, 2024-01-20:
Hi,

Thank you for the detailed post. This repository is no longer maintained. You can find the current version of WinNUT here: https://github.com/nutdotnet/WinNUT-Client

As far as LOGIN/LOGOUT is concerned, I believe WinNUT has been doing this for awhile now. Please give the latest version a try, and if you're still experiencing problems, please make a post in our Q&A discussion area and we can go from there.

whc2001, 2024-01-20:
Ahhhh never realized there is another fork. I'll check it later, thanks a lot!

closed 2024-01-20 by whc2001