gawindx/WinNUT-Client #189: wrong UPS load with APC Smart UPS 1000, DS220+

jameshlog, 2024-05-07:
Hi,

with the windows client i got a max load on the "UPS Load" indicator of 0 W, the UPS is able to handle 600 W / 1000 VA.
I have windows 10 and the last version of Winnut (2.2.8719.24624)
DiskStation DS220+ is Last firmware DSM 7.2.1-69057 Update 5
APC Smart UPS 1000 (model SMC1000IC) and firmware 02.0 (1061)

[image]

[image]

[image]

here my UPS Variables

ups (American Power Conversion/Smart-UPS_1000/UPS 02.0 / ID=1061)
battery.charge (Description unavailable) : 100
battery.charge.low (Description unavailable) : 10
battery.charge.warning (Description unavailable) : 50
battery.runtime (Description unavailable) : 1085
battery.runtime.low (Description unavailable) : 150
battery.type (Description unavailable) : PbAc
battery.voltage (Description unavailable) : 26.3
battery.voltage.nominal (Description unavailable) : 24.0
device.mfr (Description unavailable) : American Power Conversion
device.model (Description unavailable) : Smart-UPS_1000
device.serial (Description unavailable) : * Encryption *
device.type (Description unavailable) : ups
driver.name (Description unavailable) : usbhid-ups
driver.parameter.pollfreq (Description unavailable) : 30
driver.parameter.pollinterval (Description unavailable) : 5
driver.parameter.port (Description unavailable) : auto
driver.parameter.synchronous (Description unavailable) : no
driver.version (Description unavailable) : DSM7-2-1-NewModel-repack-64570-230831
driver.version.data (Description unavailable) : APC HID 0.96
driver.version.internal (Description unavailable) : 0.41
ups.beeper.status (Description unavailable) : enabled
ups.delay.shutdown (Description unavailable) : 20
ups.firmware (Description unavailable) : UPS 02.0 / ID=1061
ups.mfr (Description unavailable) : American Power Conversion
ups.mfr.date (Description unavailable) : 2023/10/19
ups.model (Description unavailable) : Smart-UPS_1000
ups.productid (Description unavailable) : 0003
ups.serial (Description unavailable) : * Encryption processing *
ups.status (Description unavailable) : OL
ups.timer.reboot (Description unavailable) : -1
ups.timer.shutdown (Description unavailable) : -1
ups.vendorid (Description unavailable) : 051d

gbakeman, 2024-05-07:
Hi @jameshlog ,

It looks like you're using a newer release of WinNUT, but you somehow made it back to the old repository! Please first download the latest pre-release by either changing your **Update Channel** setting to `Development`, or download the latest update manually [here](https://github.com/nutdotnet/WinNUT-Client/releases). If that doesn't change anything, please see [this](https://github.com/nutdotnet/WinNUT-Client/issues/122) issue for a similar problem you're experiencing, and if nothing else helps there, go ahead and add your comment to it. Just by looking at your very informative post here (thank you!), I don't think we're going to be able to get the information you want from the NUT server. It's providing only information regarding the battery, and not input, output or load & power stats that we need. Please also see [here](https://github.com/nutdotnet/WinNUT-Client/discussions/144) regarding missing variables.

jameshlog, 2024-05-07:
I tried updating to the latest version in the link below.
link : https://github.com/nutdotnet/WinNUT-Client/releases/tag/v2.3.8892

but, there was no change from before the update.
My UPS is connected to synology DS220+ via USB Port.

UPS Variables is same.

[image]

gbakeman, 2024-05-07:
It does seem like the issue lies somewhere in between the UPS and the NUT server then. You can check [here](https://networkupstools.org/stable-hcl.html), but I don't think I see your exact UPS on the hardware compatibility list. Let's close down the issue here and move discussion over to the new WinNUT repository. If you'd like additional help with the NUT server, please feel free to create a new discussion [here](https://github.com/nutdotnet/WinNUT-Client/discussions/categories/q-a).

itgmkol, 2024-08-14:
I have the same problem on qnap. Do you have a solution?

gbakeman, 2024-08-14:
@itgmkol I want to point out that this repository is no longer maintained, and development is [over here now](https://github.com/nutdotnet/WinNUT-Client). You'll want to make sure you're running the latest release of WinNUT from there, then take a look at [this discussion topic](https://github.com/nutdotnet/WinNUT-Client/discussions/155) where @jameshlog posted about a similar UPS model. Hope some of this helps!