gawindx/WinNUT-Client #190: WinNUT Client installed but not connecting to UPS via NAS

cbturner46, 2024-05-30:
I have connected my APC UPS to my Synology NAS and configured the NAS with enabled USB UPS support and enabled network UPS server with permitted device IP for my Windows 11 PC (fixed IP address).  The NAS sees my UPS ok and shows its status, state of charge etc.
I installed the latest WinNUT-Setup.msi into default (x86) location and started it ok.  Went to setup and and put in the IP address of my NAS (again fixed), left port at default 3493 and login credentials as upsmon and secret.  I also tried monuser.  Tried leaving polling interval at 0, also tried it at 1.
Following this, the NUT Client display does not show my UPS information, online, voltage, charge etc. - nothing there at all.
So what am I missing in the setup process?
I notice that the Client reports Unknown UPS Name each time I edit the Settings dialogue.  Clicking Reconnect doesn't help.
I'm new to WinNUT and just a simple Windows user.  Thanks for any help.

cbturner46, 2024-05-30:
My NUT Client is version 2.0.7722.30975

gbakeman, 2024-05-30:
Hi,

Please see #136. If updating to the latest WinNUT doesn't help, please open a new [discussion](https://github.com/nutdotnet/WinNUT-Client/discussions) and I'll see if I can assist.